RehabMan/patch-nvme #3: NVME drive not showing up

EricTheRed1, 2016-07-22:
I need some help.  I just downgraded from the Sierra public beta where I had the NVME patch working just fine.  Now i'm in 10.11.6 and nothing happens when I install the kext.  The NVME drive does not show in disk utility and the system report now shows no NVME drive detected.  

I get no errors when creating the kext file and I'm installing it just the same as I did on Sierra.  I even re-built my config.plist file before creating the kext with no luck.  

Thanks for helping

RehabMan, 2016-07-26:
Creating the kext for 10.11.6 requires different usage.  Make sure you read the README.

closed 2016-07-26 by RehabMan

EricTheRed1, 2016-07-26:
We were discussing this issue on the tonymacx86.com forums.  I am using the correct commands to create the kext.  The 10.11.6 kext is not working.  I have tried the old NvmeGeneric kext and it is working right now so it's not the drive.

RehabMan, 2016-07-27:
Problem then is compatibility of Pike's patches when coupled with your hardware.