RehabMan/patch-nvme #5: Poor 10.12.0 Performance

maleorderbride, 2016-09-22:
I have installed the 10.12.0 version of HackrNVMeFamily (I had to put in SLE to get it to load), but I have very slow NVMe performance with my Intel P3605 PCI-e SSD.

I am getting 645/640 write/read with AJA System Test versus north of 1600/1600 under 10.11.6 with GenericNVMeFamily.kext.

Any ideas? What files can I provide?

RehabMan, 2016-09-22:
No idea.  The patches are done based on Pike's work.

closed 2016-09-22 by RehabMan

hugheba, 2016-09-28:
I'm in the same situation but I am using a Samsung  950 Pro M2 drive only getting about 40 read/write.

I was using the NVMeGeneric.kext getting around 1400 read/write in El Capitan. 

Will follow this thread to see if anyone has any ideas.

NickWoodhams, 2016-09-28:
@hugheba Do you still have any other NVMe kexts installed? Run "sudo kextstat | grep NVM" in terminal and check the output. I am running 10.12.0 on a raid 0 stripe of 2 950 Pros and getting solid read/write speeds.. even though there is an error spamming my console. 
[image]

hugheba, 2016-09-28:
@NickWoodhams I've made sure to remove the NVMeGeneric.kext I was using for El Capitan cause Sierra would kernel panic with it.

% sudo kextstat | grep NVM
Password:
   56    0 0xffffff7f81d4d000 0x38000    0x38000    com.apple.iokit.IONVMeFamily (2.1.0) B460E2CE-D16E-3C0F-B41F-0E2A84FEB291 <42 27 24 23 12 7 6 5 4 3 1>

I do notice something strange which is that I now have a new virtual drive /dev/disk4. It looks to maybe be part of the new APFS.

Disk Utility lists /dev/disk4 as the mount for "Macintosh HD" which is the drive that is performing slowly, both in Aja and Black Magick Disk Speed Test.

[image]

[image]

[image]

RehabMan, 2016-09-29:
If you're following the README here, it would be listed as com.apple.hack.HackrNVMeFamily, not com.apple.iokit.IONVMeFamily

Clearly you didn't follow the README.