RehabMan/patch-nvme #6: ApplleNVMe Assert Failed: 0 == (status)

NickWoodhams, 2016-09-27:
Generated the HackrNVM kext for Sierra using your latest update. What's strange is that my NVMe drives load but this message is continually spammed on my logs. Unloading and reloading the kext on an AHCI drive didn't get rid of the error. Any ideas?

I have 2 950Pros, 1 SM951 (AHCI), and 1 XP941 (AHCI)

default	22:43:25.969568 -0400	kernel	AppleNVMe Assert failed: 0 == (status)
default	22:43:25.969576 -0400	kernel	Exit
default	22:43:25.969578 -0400	kernel	file: /BuildRoot/Library/Caches/com.apple.xbs/Sources/IONVMeFamily/IONVMeFamily-234.1.16/AppleNVMeRequest.cpp
default	22:43:25.969584 -0400	kernel	line: 690

[image]

RehabMan, 2016-09-27:
As per README, you must remove IONVMeFamily.kext from /S/L/E.
Always read carefully.

closed 2016-09-27 by RehabMan

NickWoodhams, 2016-09-27:
@RehabMan I appreciate your hard work. I did read the entire README carefully. I previously removed that kext per your README. I also rebuilt the cache and tried loading without caches as well. Loading HackrNVM with Clover injection, does the same thing when the patched kext is installed to S/L/E
[image]

RehabMan, 2016-09-27:
The error in syslog is probably from a prior time when you had IONVMeFamily still present.