RehabMan/patch-nvme #8: macOS Sierra Hackintosh

snapo, 2016-10-25:
First of all thanks for creating those patches... i did just try to install macOS Sierra on a Lenovo T460s with a pci express SSD. Clover recognizes the disk without an issue... but macOS Sierra not (On the install screen i only see the bootable USB Stick).
I did try to run it with your kext files and put them in EFI/Clover/kext/other ... but as soon as i add them macOS Sierra will not boot the installer and exit with a kernel panic. Is there any information i can provide you that would help in fixing this or can i somehow solve it by myself?
My guess is that the NVM SSD has another ID ... because it is directly from Lenovo.

When i remove the generated kext the installer works again....

I did try the 10_12_0 and 10_12_dp1 , same result on both

[image]

hkoivuneva, 2016-10-27:
@snapo if you didn't yet try it the way RehabMan instructed, I hope this helps :)

From the readme:

> **# 10.12 Sierra Notes**
> 
> With 10.12 there are a couple of procedural changes:
> 
> If you are trying to use HackrNVMeFamily for the 10.12 _installer_, forget about it. **Use the correct patches in config.plist KextsToPatch**.
> 
> Once you install, you can create the HackrNVMeFamily and use it (or use one you already created), but you must remove IONVMeFamily.kext from /System/Library/Extensions
> 
> I will update here when/if there is a better solution.

-------------

Pikeralpha post: https://pikeralpha.wordpress.com/2016/06/27/nvmefamily-kext-bin-patch-data
Link to instruction post: http://www.insanelymac.com/forum/topic/312803-patch-for-using-nvme-under-macos-sierra-is-ready/
Here's the IONVMeFamily.kext mentioned in the thread, I extracted it just now: [https://goo.gl/7zeFcC](https://goo.gl/5svg9E)
(My backup folder)

snapo, 2016-11-02:
Thanks, i did try that already. Clover see's the NVM SSD but not the ossierra installer. Will try it again this evening and build everything from scratch again.

RehabMan, 2016-11-23:
READ the README.

closed 2016-11-23 by RehabMan

ShikiSuen, 2016-11-24:
To OP: Please do post your entire XML content of your config.plist here plus your hardware configuration (logic board model number, graphics model number, CPU model number, etc. just in case). Only in this way other people could see what had happened with your clover configuration. Problems could happen if your applied patch doesn't match your system build.