RehabMan/patch-nvme #13: not working with Intel 600p NVMe

zerog2k, 2017-01-14:
Hi @RehabMan 

I have Intel nvme m.2
SSDPEKKW512G7

I have tried using patch-nvme, then tried using method outlined here with SSDT patch, with spoof kext, and also rename _DSM->XDSM patch.
https://www.tonymacx86.com/threads/guide-hackrnvmefamily-co-existence-with-ionvmefamily-using-class-code-spoof.210316/

Still no luck. SSD not visible in OSX.

    | |   |         "compatible" = <"pci8086,390a","pci8086,f1a5","pciclass,010802","PXSX">

Reading in some other forums where you were helping, other users with these Intel SSDs might be having similar issue:
http://www.insanelymac.com/forum/topic/312803-patch-for-using-nvme-under-macos-sierra-is-ready/page-38

Wondering how to go about getting this working.

ShikiSuen, 2017-01-15:
I only use clover hot patch and my Intel 600p (1TB) works fine.
Looks like my preference of reducing non-genuine Apple genuine kexts really made some benefits in my case.

RehabMan, 2017-01-15:
This is not the appropriate place for troubleshooting.
Two possibilities:
- you made a mistake
- Pike's NVMe patches are not compatible with your device

Use the forums.

closed 2017-01-15 by RehabMan

ShikiSuen, 2017-01-16:
@zerog2k Once you asked the same troubleshooting request at any forum, please leave your thread link(s) here for the sake of guiding other visitors here to the ideal place.

RehabMan, 2017-01-16:
READ the README

ShikiSuen, 2017-01-24:
@RehabMan I just tested on 10.12.2 using the kext generated by your script:

<pre>
mkdir ~/Projects && cd Projects
git clone https://github.com/RehabMan/patch-nvme.git patch-nvme.git
cd patch-nvme.git
./patch_nvme.sh --spoof 10_12_2</pre>

The kext, wherever it gets placed (Clover kext folder or /Library/Extensions), just let my system failed to pass the DSMOS (a disabled icon on screen in lieu of the white Apple logo).

I confirmed that none of my steps are wrong. Pike's patch worked with my Intel 600s well if applied in config.plist instead of generated HackrNVMeFamily kext.

RehabMan, 2017-01-24:
To use --spoof generated kext, you need to inject the spoofed class-code.
READ the README and related links.