RehabMan/patch-nvme #19: Causing Kernel Panick in Hackintosh 10.12.3

rowild, 2017-02-08:
Hi, 

I tried to use your script on the above mentioned system (specs below, running stable) to get this PCIe card running:
http://www.samsung.com/semiconductor/products/flash-storage/client-ssd/
http://geizhals.at/samsung-ssd-pm961-1tb-mzvlw1t0hmlh-00000-a1486269.html

The PCIe card is recognised in the BIOS, but nowhere within mac OS X Sierra (no DiskUtility etc).

Following various threads on tonymacx86, I found this script, followed the process to generate a Hack kext for 10.12.3 and installed it using "Kext Wizard" (and corrected permissions). After reboot, the card is still not recognised by OSX. And after 2, 3 more reboots, a kernel panic disrupts the boot process of the system.

I remove the kernel panic by following this guide (unfortunately I do not know how to hold the screen at that point, so I cannot post a screenshot; all camera images are blurry... but it does mention "IOUSBFamily" in the track back, which I also need to remove first, see below):
https://www.tonymacx86.com/threads/guide-remove-extra-clover-bios-boot-entries-prevent-further-problems.175274/

- Loading system from USB
- Start Clover Boot Loader
- Start Clover UEFI shell
- Remove not needed partitions

Then I start OSX from USB, cd into /Volumes/System/Library/Extensions/ on the Hackintosh built and remove the Hack kext AND the above mentioned IOUSBFamily.kext (by renaming the latter to 'IOUSBFamily.kext.bck').

After that, the system can be restarted again. I also rename the IOUSBFamily.kext.bck to its original name again, and everything keeps working.

This is repeatable, but I never checked, if it is really necessary to rename the IOUSBFamily.kext, to be honest. I will next time and update my post here. Promised :-)

(However: I still don't know, how to activate my PCIe card... which is not the primary intend of posting this issue, though...)

My Hackintosh specs:
Gigabyte Z170x UD5 TH
Intel i7 6700K
NVidia 970 Windforce
SanDisk Ultra II 960GB
32 GB RAM

I do not know if this is of any use. If not, just ignore. 
Thank you!

RehabMan, 2017-02-08:
It is clear you didn't follow README instructions...
See class-code spoof guide linked by README.

closed 2017-02-08 by RehabMan

rowild, 2017-02-08:
I followed those instructions exactly and read through tons of threads. How come you say I didn't follow the instructions clearly?

RehabMan, 2017-02-08:
Your responsibility to read carefully.
Use the class-code spoof.
Guide is linked by the README.