RehabMan/patch-nvme #20: Get panic

diepnt90, 2017-03-02:
Dear RehabMan,
 
im using IONVMeFamily.kext  and code add to KextstoPatch as #8. but sometimes, my hackintosh dead, it said the osx partition is not verified completely.
 
yesterday, i tried to use nvme_patch on your github but got panic. could you help to check my steps below is true or faile
 
1. reinstall mac os 10.12.3 using IONVMeFamily.kext and add code to KextstoPatch
2. after installing successfully, I download, run and make HackrNVMeFamily-10_12_3.kext
3. remote IONVMeFamily.kext in SLE, remove any <dict>...<\dict> related with IONVMeFamily.kext
4. using kext ward to install HackrNVMeFamily-10_12_3.kext
5. reboot and get panic. tried to boot using -v but the text show on sreen is hard to read. 
 
im using OCZ SSD nvme
thanks.

RehabMan, 2017-03-02:
It is clear you didn't read the README.
Read it carefully.

closed 2017-03-02 by RehabMan

ShikiSuen, 2017-03-06:
Using HackrNVMeFamily kext, if not --spoof generated, requires the absense of Apple official NVMe kext. Otherwise, a kernel panic on startup is expected.

If --spoof generated, you need to inject the spoofed class-code.

In my case I kept on using clover configurations. I know the risk addressed in README and I constantly use TimeMachine to backup my data to a mechanical hard drive.