RehabMan/patch-nvme #21: Sierra 10.12.3 and Samsung 950 PRO

alex88, 2017-03-25:
On El capitan I've used my M2 disk with the generic NVME driver, now with sierra I was trying to use yours.

When you say

> If you are trying to use HackrNVMeFamily for the 10.12 installer, forget about it. Use the correct patches in config.plist KextsToPatch

"the correct patches" are? the one for the OS I want to install? Because from the default unibeast config.plist I've tried to add the dicts from https://github.com/RehabMan/patch-nvme/blob/master/NVMe_patches_10_12_3.plist however

[image]

RehabMan, 2017-03-25:
| "the correct patches" are? the one for the OS I want to install?

Yes.  See the various plist files in the repo.
But you should use the class-code spoof anyway...

| now instead I get still waiting for root device when booting from original disk

Need same patches on your EFI/Clover after you install.
Suggestion: use the class-code spoof...

| Update 2: after building the kext with --spoof I get the another kernel panic

Must read the spoof guide carefully.

| Update 3: I've removed any patched kext, just applied the kext patch into clover config.plist and it worked with vanilla IONVMeFamily kext

It is fine, but updates can be complicated (the patches required pre-update and post-update may be different).

You should use the class-code spoof.
Read the README and related links carefully.

closed 2017-03-25 by RehabMan

alex88, 2017-03-25:
Oh thank you very much for answering.. I thought the links in the spoof section were just in case I wanted some details on the implementations, not mandatory to have it working.. Thanks

RehabMan, 2017-03-25:
There are many ways to "have it working".  Using the class-code spoof is the best way.