RehabMan/patch-nvme #22: Long story with Hackr on original iMac

noobissat, 2017-03-28:
Dear RehabMan,

Greetings from Russia. I would be happy if you have several minutes to answer my questions. I am among few people in the country who tried to adapt third party M2 SSD to original Mac using your patched kext. At the moment all efforts seem to be unsuccessful. 
Since all available comments and explanations pay more attention to Hackintosh configurations (namely, Clover), I'll try to clarify some issues about original Mac.

First, I own iMac 17.1 (aka Retina 5K late 2015) with Sierra 10.12.3 (also I’ve recently updated system to 10.12.4 beta 8 and tried with it as well) and Bootcamp installed Windows 10 running on it. And I installed Samsung 960Pro via custom-made adapter. 

Second, Windows correctly recognized SSD, was able to format it, and performance tests were all done successfully shown results similar to that of in popular reviews. So I concluded that adapter works properly.

Remarkably, under macOS (both 10.12.3 and 4 beta8) vanilla IONMVeFamily.kext showed some activity. Namely, in NVMe chapter of standard system configuration (about this Mac) one can see Apple NVMe adapter device list, but it is empty (pic.1).

[image]

 ACPI explorer shows disc with its right name and s/n (pic.2).

[image]

Third, I tried to apply the patch according your manual (https://github.com/RehabMan/patch-nvme), also I tried to use already patched kext (downloaded from internet), also I changed the name of patched kext to the standard IONVMeFamily and installed it instead of vanilla. I even used patched binary with the original info.plist.

How did I prepare OS:
1) turned off SIP (through command+R and terminal/csrutil disable);
2) made kextdevmode=1;
How did I install/remove kexts:
1) Mainly using KEXTutility.
2) Using Terminal as well (several times I repeated all the actions earlier made with KEXTutility using Terminal).
3) I tried installations with or without spoof command (before the moment when I realized that spoof is useless in original Mac).
4) I tried with and without prelinked kernel refresh, with and without kext cache rebuilt, etc.
All these manipulations always led to two results. Kext either didn’t work at all (showing nothing in ACPI space) or caused panics. I caught at least 3 sorts of kernel panics during this period (pics 3-5).

[3.rtf.docx](https://github.com/RehabMan/patch-nvme/files/876850/3.rtf.docx)

[4.rtf.docx](https://github.com/RehabMan/patch-nvme/files/876851/4.rtf.docx)

[5.rtf.docx](https://github.com/RehabMan/patch-nvme/files/876852/5.rtf.docx)

Also I tried to edit vanilla IONVMeFamily with Hex-editor (I mean changing preferable block size). In all 4 cases disc remained visible in ACPI explorer, but not accessible in Disc Utility and System configuration.

Finally, my questions on all this:

1) Does the original Mac require some additional manipulations (such as bootloader modification, etc) to use HackrNVMeFamily.kext?
2) Are there any perspectives to use third party M2 SSD as a bootable device in original Mac?
3) What is a possible explanation of my failure with patched kext?
4) Maybe it is reasonable to patch vanilla kext partially, in Hex-editor, in my case?

Great thanks in advance and sorry for my poor English!

Sincerely,

Alex

RehabMan, 2017-03-29:
Since you can't inject the spoofed class-code, you cannot use --spoof generated kext on a real Mac.
And if you plan to use HackrNVMeFamily (no spoof) on 10.12.x, you must remove IONVMeFamily, as per README.

closed 2017-03-29 by RehabMan

noobissat, 2017-03-29:
Good morning!

Of course in the case of no-spoof usage I removed IONVMeFamily. HackrNVMe caused panic (for example, like in 3.rtf.docx).

RehabMan, 2017-03-30:
You probably patched with the wrong patch file.
Make sure the md5 sums matched (the patch_nmve.sh script tells you).
If you don't have matching md5 (expected and patched), then you're doing something wrong.

noobissat, 2017-03-30:
Hi,

MD5 sums were both correct (in 10.12.3 as well as in 10.12.4 beta8). I checked every step in a strict accordance with the Readme.

One more detail. I also tried to check NVMeGeneric kext (of course first I removed IONVMeFamily), and it caused panic too.

RehabMan, 2017-03-30:
No other ideas for you.

noobissat, 2017-03-30:
Anyway, thank you so much for response. At least your statistics on typical mistakes with this patch is really wide, and I'll for sure check everything again.

PS: Interestingly, I've found several posts from another real-Mac users with the similar situation in Picke R. Alpha blog. Maybe there is a common with problem with generic NVMe usage originating from some real-Mac hardware features...

RehabMan, 2017-03-30:
| PS: Interestingly, I've found several posts from another real-Mac users with the similar situation in Picke R. Alpha blog. Maybe there is a common with problem with generic NVMe usage originating from some real-Mac hardware features...

Sounds likely.

noobissat, 2017-04-03:
Dear Rehabman,

I found the combination that works for me. It is the 'minimal' patch applied to vanilla kext using Hex-editor

(find <0fb68c88 82000000 83f90c0f 85320100>
repl <0fb68c88 82000000 83f9090f 85320100>

find <00c783a4 00000000 10000048 8b480848>
repl <00c783a4 00000000 02000048 8b480848>

find <89c181e1 ff0f0000 4801d148 81f9ff0f 0000773b>
repl <89c181e1 ff010000 4801d148 81f9ff01 0000773b>

find <4981c5ff 0f000049 c1ed0c49 8b042448>
Real <4981c5ff 0f000049 c1ed0949 8b042448>)

However, previously I read your warnings about partially patched kexts. I intend to find the particular bytes which cause panic. Could you please provide me with the Hex codes for the complete set of patches. Sorry for stupidity, but my attempts to obtain these Hexs from Base64 info (using online decoders) completely failed(

Great thanks in advance