RehabMan/patch-nvme #25: Vanilla MD5 does not match

adarwash, 2017-05-20:
Every time i run script i keep getting this error:

Creating patched HackrNVMeFamily-10_12_4.kext
WARNING: Vanilla MD5 (8a4cdcb003221e8661299d300805dba3) does not match expected MD5 (cd81a56a2baa193cc1c66e8b52a618e7)
ERROR: Vanilla MD5 does not match and --override not specified.  No kext generated!

This is the command i am writing : 
./patch_nvme.sh 10_12_4

RehabMan, 2017-05-20:
The md5 8a4cdcb003221e8661299d300805dba3 does not match any known md5 values for any version of macOS/OS X.

Your IONVMeFamily.kext is corrupt, or already inappropriately patched.
Use the macOS installer to refresh all your kexts.

closed 2017-05-20 by RehabMan

vizualview, 2017-11-11:
RehabMan 

Thanks for your awesomeness! Can you help me? I'm having this similar problem. I tried reinstalling El Capitan 10.11.6. I tried using Kextutility. No luck. I'm getting Kernal panics on shutdown and my machine reboots. So it been on for days. Hoping this patch will fix. How can I refresh the kexts?

Vanilla MD5 matches expected MD5 entry (1ccd3c76e5fc8a952baccabbc57e879f)
Patched MD5 matches expected MD5 entry (c2c7d18760a71d28e7857955cdcfe30e)

RehabMan, 2017-11-11:
Sounds like your issue is unrelated.
Use the forums.

feinberj-github, 2018-10-29:
I was having this problem on my 5,1 2009 Mac Pro, with a clean build of 10.12.6 with the latest udpates installed.  The problem was this...  I had installed the Mojave beta on this machine, on a different drive.  **This caused my machine to be registed in the public beta program**, and as such when I installed Sierra it then updated itself with PUBLIC BETA security updates and not the regular security updates.  As such the MD5 key calculated from my public beta kext was not matching anything expected by the patch NVMe script.  This was particularly maddening, since I installed Sierra using an unaltered installer, let the updates run, then tried running this script.  I figured out what was going on when I ran About this Mac->System Report->Software Installations and saw all of the public beta security updates in there.    I have since removed my machine from the public beta program  :)

banjo1000, 2020-12-10:
MacBook Air 6,2 MacBook Air (Mid-2013)
Sintech adapter card, XPG SX8200 Pro (2TB)
MacOS Sierra 10.12.6 (16G2136)
I was never part of Mojave beta program, but I have installed Mojave previously on this laptop, if relevant.

I am booting off external SSD. I want this OS to see my internal SSD, and I eventually want to install Sierra on internal SSD, but the Sierra installer doesn't see the internal SSD.

Here's what I've tried while booted in Sierra from my external SSD:

Joes-Air:patch-nvme-master joe$ ./patch_nvme.sh 10_12_6
Creating patched HackrNVMeFamily-10_12_6.kext from /System/Library/Extensions/IONVMeFamily.kext
WARNING: Vanilla MD5 (da9935834b84e98f05afe4063d964635) does not match expected MD5 (c506f1fc40026c0262a736f0be318223)
ERROR: Vanilla MD5 does not match and --override not specified. No kext generated!
Joes-Air:patch-nvme-master joe$

Should I expect patch-nvme-master to work with my SSD which isn't enumerated in the script? If not, can I edit the script to fix? Should I use the --override option?

How to get around the limitation that when booting off my bootable USB Sierra installer, the installer can't see my internal SSD or an APFS volume on it where I want the new Sierra OS to be?

Thanks.