RehabMan/patch-nvme #27: 10.12.5 and Samsung 960 EVO Issue

quinton-ashley, 2017-05-29:
I recently had a working setup with 10.12.4 and the Samsung 960 EVO which I was not using as my boot drive.  I was using your patch and it worked really well!  When I updated to 10.12.5 my boot drive became unbootable and an error log appeared on the screen that said IOAHCIFamily and IOStorageFamily were in backtrace.  I think I screwed up and should've put the vanilla IONVMeFamily.kext back before updating right?  I erased and reinstalled on my boot drive and now it's running 10.12.5.  I followed the instructions on the README and rebooted a few times but my 960 still isn't showing up.  What should I do?

RehabMan, 2017-05-29:
You clearly neglected to implement the class-code spoof.
Guide is linked from the README.

closed 2017-05-29 by RehabMan

quinton-ashley, 2017-05-29:
Should I be keeping IONVMeFamily in /S/L/E?  I thought the --spoof option did the class code spoofing automatically?

RehabMan, 2017-05-29:
If you completely implement the class-code spoof, you can leave IONVMeFamily.kext in /S/L/E.
If you don't implement spoofing, you need to remove it (which presents an issue with updates).
Implementing the class-code spoof involves more than just --spoof (requires also correctly coded class-code injection SSDT, and removal/renaming of the _DSM at that scope).  It is all covered in the guide.

quinton-ashley, 2017-05-30:
Should I not use the spoof option if I don't want to implement the class-code spoof?  I looked at the guide and I'm sure I didn't do that last time.

RehabMan, 2017-05-30:
The --spoof option can only be used if you're implementing the class-code inject via SSDT.
If you generate a --spoof kext without the SSDT, it will never load.
You should implement the class-code spoof, or you'll have panic next update...