RehabMan/patch-nvme #33: Issue after installing Security Update 2007-001 on Sierra 10.12.6

jecor, 2017-11-02:
It seems the update installs a new version of IONVMeFamily, which results in a kernel panic upon reboot because of a conflict with HackrNVMeFamily:

OSMetaClass: Kext com.apple.iokit.IOVNMeFamily class IONVMeBlockStorageDevice is a duplicate;kext com.apple.hack.HackrNVMeFamily already has a class with that name.

With trying to patch the "new" IONVMeFamily, it does not work because the kext changed:

./patch_nvme.sh 10_12_6 

Creating patched HackrNVMeFamily-10_12_6.kext from /System/Library/Extensions/IONVMeFamily.kext
WARNING: Vanilla MD5 (a0bbf560c4792dfa5d569f7ec6eef096) does not match expected MD5 (c506f1fc40026c0262a736f0be318223)
ERROR: Vanilla MD5 does not match and --override not specified.  No kext generated!

It is likely the OSX build number also changed: 16G1036, since NVidia drivers needed a new install.

ShikiSuen, 2017-11-04:
I am afraid that macOS 10.13 may no longer need the NVMe patch in this repository in most cases.
I patched nothing from this repository and my Brix runs well with macOS 10.13 and my two NVMe SSDs: Intel 600p and SAMSUNG 950 Pro.

jecor, 2017-11-06:
Hi ShikiSuen, thanks for your reply, but isn't it to issue #34? This post is for Sierra (10.12), not High Sierra :-)

ShikiSuen, 2017-11-07:
@jecor Thanks for mentioning. My apologies for my misreading of your posts.
Still you could share your concerns here regarding why you have to use macOS 10.12 in lieu of 10.13 at this moment, and I guess I could solve your concerns. (This is out of topic, though.)

RehabMan, 2017-11-08:
By design.

This why you're supposed to use the class-code spoof.
The fact that HackrNVMeFamily (non-spoof) does not co-exist with IONVMeFamily is documented in the README.

closed 2017-11-08 by RehabMan

jecor, 2017-11-19:
@RehabMan: total agreement: non-coexistence is documented and I could use a class-code spoof. I was actually just asking for a proper patch for the security update variant, which you have done now in 458711e6d76d571ae11ca0d6ff9d81f04013442f. Thanks a lot :-)

@ShikiSuen: agreed, 10.13 should solve my problem. However, given the forums reporting many problems with NVidia drivers install and some purely Apple problems (iMessage not behaving correctly even on actual Macs), I was not very keen on upgrading immediately. Thanks for the advice anyway.

ShikiSuen, 2017-11-23:
@jecor iMessage behaves as normal on my side. As long as every sub-parameter of your Clover SMBIOS identifiers is configured correctly and unique, you shouldn't have iMessage problems (unless you previously met too much problems with one Apple ID, and Apple may force you to change your Apple ID password).

Regarding NVidia GPU issues, see:
http://www.macvidcards.com/drivers.html
As long as you install driver at least 378.10.10.10.15.121, you shouldn't meet the GPU issues you mentioned above. Even if you met the issue, you could temporarily cancel the webdriver load from your Clover and get into the system to finish the installation process then remove the old video driver.

ShikiSuen, 2017-11-23:
Sorry for mistyping in my last reply. The usable High Sierra NVidia Webdriver shall not be earlier than 378.10.10.10.20.107.