RehabMan/patch-nvme #34: 10.13.1 IONVMeFamily No  f6c1100f 85410100 00

935713827, 2017-11-04:
10.13.1 IONVMeFamily No  f6c1100f 85410100 00

ShikiSuen, 2017-11-07:
What did you patched? (Sorry, I have to ask this question just in case.)

ShikiSuen, 2017-11-07:
Also, please provide the following information:
1. the exact model of the SSD you are using.
2. (supplementing what I asked in my previous reply) Please describe what you did in detail regarding your patching process.

P.S.: I am afraid that macOS 10.13 may no longer need the NVMe patch in this repository in most cases. I patched nothing from this repository and my Brix runs well with macOS 10.13.x and my two NVMe SSDs: Intel 600p and SAMSUNG 950 Pro.

RehabMan, 2017-11-08:
No need for this script in 10.13.x.  The native kext can now handle 512 byte LBA.

closed 2017-11-08 by RehabMan