RehabMan/patch-nvme #35: Recovery

asheenlevrai, 2017-11-10:
Hello :)

I want to boot into recovery in order to clone my AHCI sATA SSD onto my (now visible thanks to your patch) NVMe M.2 SSD. However in recovery mode, disk utility cannot see the NVMe drive since the necessary kexts are missing (In Clover? In EFI/Clover/kexts/Others?).

Where and what kexts should I copy in order to be able to see the NVMe drive in disk utility while booted into the recovery mode?

Thank you very much in advance for your help.

Best,
-a-

RehabMan, 2017-11-10:
You must implement the class-code spoof.
The guide is linked from the README.

closed 2017-11-10 by RehabMan

ShikiSuen, 2017-11-12:
or upgrade to macOS High Sierra and later.

RehabMan, 2017-11-12:
For the most part, the content of this github project does not apply to 10.13.x and later.
Read the README for details.

ShikiSuen, 2017-11-14:
@RehabMan Yeah. I am suggesting him to upgrade the system so he won't be able to bother you again ('cause of what you said above).