RehabMan/patch-nvme #36: 10.12.6.Security Update 2017-002

asheenlevrai, 2017-12-21:
What should be done regarding this update?

There is no corresponding .plist file available, yet in order to generate the kext file

Tx

Best,
-a-

RehabMan, 2017-12-21:
I'll check if I have 10.12.6 running anywhere so I can get native IONVMeFamily.
Likely the same set of patches can be used, but the md5 values just changed.
Or you could upload IONVMeFamily.kext from that release.

RehabMan, 2017-12-22:
There is already an existing plist for this IONVMeFamily.
NVMe_patches_10_12_6_16G1114.plist

closed 2017-12-22 by RehabMan

asheenlevrai, 2017-12-22:
thanks :)

What does 16G1114 corresponds to?

asheenlevrai, 2017-12-22:
How can I apply the patch now that the computer won't boot?

Can I do this on another computer if I add the drive as a secondary drive?
Do I just need to copy the newly created kext into SLE (and remove the original IONVMeFamily.kext)? Will that be enough?

Tx again,
-a-

RehabMan, 2017-12-22:
| What does 16G1114 corresponds to?

The build#.
Don't forget that that patch-nvme.sh script automatically chooses the appropriate plist based on md5 of native IONVMeFamily.kext in /S/L/E.

| How can I apply the patch now that the computer won't boot?

Sounds like you didn't implement the class-code spoof.
This guide is linked from the README:
https://www.tonymacx86.com/threads/guide-hackrnvmefamily-co-existence-with-ionvmefamily-using-class-code-spoof.210316/

asheenlevrai, 2017-12-23:
Yeah... Sorry but I followed [this tutorial](https://nickwoodhams.com/getting-your-nvme-ssd-working-on-macos-sierra-10-12-2/) (which I was able to understand, or so I thought), in which one needs to specify what .plist file to use. (I adapted the command in order to use the version of the plist file 10.12.16_2017-001 back then)
I didn't know the md5 was checked although the output of the CLI mentions the md5 at some point...

I am now currently reading your guide on Tonymacx86. I don't know what a _class code_ is. I don't know precisely what SSDTs are and how to handle them. Editing Clover's config.plist seems less complicated to me but your guide doesn't recommend it. Also I don't meet the following requirements:

> Assumptions:
> - installation of Windows 10 on the target
> - basic config.plist editing skills
> - basic MaciASL usage
> - basic knowledge of installing/injecting kexts

Am I screwed?

My machine only runs 10.12.6_2017-002 (actually won't boot now after upgrading to 2017-002), no Windows whatsoever...

If I place the newly generated HackrNVMeFamily.kext into EFI/Clover/kexts and delete IONVMeFamily.kext from S/L/E that's not gonna cut it, right? I mean up until the next upgrade at least...

I just want to know if there is now a way to boot my machine without the need to add back a regular sATA SSD and reinstall macOS...

RehabMan, 2017-12-24:
If you implement the class-code spoof (properly coded SSDT_NVMe-Pcc.aml, then inject kexts (HacrkNVMeFamily generated with --spoof), then you should be able to boot.

asheenlevrai, 2017-12-26:
Thanks

Will I be able to do that even if I don't have Win10 running on this machine?
AFAIU the guides requires Win10 in order to get the ACPI path of the drive (I have no idea what an ACPI path is exactly).

RehabMan, 2017-12-27:
Win10 is easy to install.
You can derive the path from ioreg data, but you probably don't know how to do that.

asheenlevrai, 2017-12-27:
You're right about that (no idea what ioreg data is)...

I guess with my limited abilities the quickest way to fix this is to reinstall macOS. I never keep important data on the system drive anyway.

Thank you for your time and all the best for 2018 :)
Cheers,
-a-

RehabMan, 2017-12-30:
No need to re-install.  You just need to create a USB with it configured for the class-code spoof.

asheenlevrai, 2018-01-01:
?
How is that? I don't get it...

RehabMan, 2018-01-02:
See class-code spoof guide.  You need only HackrNVMeFamily in Clover/kexts/Other and the appropriately coded SSDT.
Other than those special requirements, there are plenty of guides for how to create Clover USB sticks.