RehabMan/patch-nvme #48: Request: Samsung 970 Evo Plus

xqq, 2019-02-26:
970 Evo Plus is a recently released product by Samsung which is the updated version of 970 Evo.
The 970 Evo works well with hackintosh but, seems that the new 970 Evo Plus has something to do with it.

In my situation, The macOS 10.14.3 installation into 970 EvoPlus can never be successfully done. It will always result in a kernel panic followed by a reboot: (origin `IONVMeFamily.kext` without patches)

```bash
void AppleNVMeRequestTimer::PrintPending(): 251:QID=1 Deadline=37863793983 DW8=005A0002 DW11=00000000 DW12=00000307 DW13=00000000 DW14=00000000 DW15=00000000
: panic(cpu 0 caller 0xfffffff7f98c86172): nvme: "Fatal error occurred. (.........omitted)"
(omitted) /IONVMeFamily-387.238.1/IONVMeController.cpp:5327
```

The crash log shows that some NVMe command may became timed out. It may be a bug of 970 EvoPlus's firmware, or caused by the incompatibility within Apple's IONVMeFamily driver.

Other users are also reporting with system hang or crash reboot.

It might be able to solve through an `IONVMeFamily` patching, or not. In my opinion it may become a troublesome work, or maybe unsolvable. But hope there is anyone to investigate for this.

Related discusses:
https://www.tonymacx86.com/threads/will-the-new-samsung-970-evo-plus-drives-work-spoiler-alert-no-they-wont.270757/
https://www.reddit.com/r/hackintosh/comments/aprsbp/970_evo_plus_hackintosh/
https://www.reddit.com/r/hackintosh/comments/aufazp/freeze_with_samsung_970evo/

heyventrilo, 2019-03-07:
Same problem here, i can provide logs or any required info.

imxeno, 2019-05-11:
I would also be very grateful for a fix or workaround for this issue. Even just trying to use the disk under macOS results in kernel panic (I actually use it as a Windows drive which I mount using Paragon NTFS on macOS).

kennym, 2019-05-25:
Still having this issue... was running in circles pulling my hair until I stumbled over this.

kennym, 2019-05-25:
Who pays this RehabMan? I only see a bunch of people requesting things to be fixed, but I am pretty sure this guy does this all for free... next question - how can we pay RehabMan?

parkycai, 2019-06-21:
It's said that upgrading to the latest firmware will fix this problem;

trevjonez, 2019-06-21:
I can confirm once you update the firmware the drive works fine.
[image]

kennym, 2019-06-21:
Same here - I'm using it right now as we're speaking as my boot drive. No problems so far :-) glad Samsung fixed this.

closed 2019-07-02 by RehabMan