RehabMan/patch-nvme #50: 970 Evo (without plus)

taileron, 2019-04-30:
finally figured out to get the patched ... sec2019-001.kext loaded into 10.12.6 16G1918 (MBA 2015) deleted IONVMeFamily no KP but the drive displays MBR instead of GUID partition scheme and shows no content (CCC with working systems that boot correctly by usb) and all attempts to format the internal 970 Evo fail. 
970 Evo work perfect in this machine with Mojave so it is no hardware issue.
(Most of my software stuff still needs Sierra)

[image]
[image]

taileron, 2019-04-30:
when started in save mode sometimes the kext works and all the drives partitions are accessable in full speed

nicoh88, 2019-10-24:
Same issue here - with macOS Catalina, Samsung 970 Evo (without Plus) NVMe 1TB and iMac 21.5 4K Late 2015.

@taileron 
Do you solve the problem?