RehabMan/patch-nvme #52: Request Samsung PM981

hackinCAD, 2019-06-24:
Similar to the Samsung 970 evo plus, the PM981 is causing kp. - 
void AppleNVMeRequestTimer::PrintPending():
Sadly, unlike the 970 evo plus there is no firmware update available.
Some sort of patch for this, if possible, would be awesome!

bhagirathbhard, 2019-06-24:
SAMSUNG MZVLB512HAJQ-00000
PM981 has the same issue.

renyichan, 2019-07-30:
SAMSUNG MZVLB512HAJQ-00000 
PM981也有同样的问题。
mojave10.14.5无法安装

digitxo, 2019-08-05:
I also need PM981 - also I need mojave support!

MooeKayo, 2019-08-27:
I have used PM981 successfully on mojave.
[http://bbs.pcbeta.com/viewthread-1814806-1-1.html](http://bbs.pcbeta.com/viewthread-1814806-1-1.html)

MooeKayo, 2019-08-30:
> Hi - http://bbs.pcbeta.com/viewthread-1814806-1-1.html - you send me this link. But where can I find "paragon" thx!

It's just a backup tool: http://www.ntfsformac.cn/, so you can also replace it with other tools

sinag, 2019-10-01:
having random reboots on mojave with PM981

yaelahan, 2020-10-08:
> having random reboots on mojave with PM981

do you solve this issue ?